hneemann/Digital #837: Needed to increase MAX_LOOP_COUNTER in core/Model.java for selfmade 32 bit divider

FaberUnserzeit, 2021-09-16:
I made a 32 bit divider for fun, but to get it running with avoiding the "Logic seems to oscillate" error, I had to make my own built and increase MAX_LOOP_COUNTER in the Model class. I just appended a '0' and then it worked fine.
Maybe you like to make that constant a variable, that can be set by the user via an options-dialog in runtime?

Necessary files to reproduce the problem:
[32bitdivider.zip](https://github.com/hneemann/Digital/files/7178420/32bitdivider.zip)

hneemann, 2021-09-16:
This circuit is funny. It is the least efficient circuit I have ever seen, but it is really cool.

closed 2021-09-16 by hneemann

FaberUnserzeit, 2021-09-16:
@hneemann 
Wow, that was quick! 
As I said, I did it for fun. I am actually rather an old fashioned programmer for PCs and µCs than a specialist for digital circuits and I have no clue, how a divider ist best built. My current spleen is not to use any predefined circuits but to make everything myself bottom-up. I have derived the circuit from an algorithm Ben Eater explains on youtube. My other implementation is an 8 bit divider I made with minetest mesecons, where I made individual comperators and subtractors with different bit-sizes, thus reducing the number of gates by half, but this is the only enhancement I could think of, yet, and I found no circuit diagrams of better dividers in the web.
Btw: Digital is brilliant! I know it for less than a week and I love it! I know it thanks to rj45 Creates' video series on youtube mentioned somewhere here. - Which is hard to understand for a noob like me. :-D 

[image]

hneemann, 2021-09-17:
@FaberUnserzeit 
You might want to think about implementing a carry-look-ahead adder.  Ripple carry adders are simple to build and easy to understand, but very slow. Especially when the number of bits is large and several are connected in series. That's why they are used exclusively in teaching, but not on silicon.
And also comparators can be built following a similar idea.

See _examples/combinatorial/FullAdderCLA.dig_ and _examples/combinatorial/Comp7485Parall.dig_ for inspiration.

hneemann, 2021-09-18:
@FaberUnserzeit 
I replaced your 32-bit ripple carry adder with an adder made up of four chained 8-bit cla adders.
This alone improves performance dramatically. The divider is almost usable now. :smile: 
One would have to try out what could be gained by optimizing the comparator as well.

[fullAdder32.zip](https://github.com/hneemann/Digital/files/7189488/fullAdder32.zip)

hneemann, 2021-09-18:
@FaberUnserzeit 
I played around a bit more with the divider: :smile:
With the original divider, there are a maximum of 1781 gates between one of the inputs and one of the outputs.
If the 8 bit CLA adder is used and also the comparator is optimized similar to the 74x85, there are at most 81 gates in series. This corresponds to a factor of 22 in runtime. In addition, the faster design also requires somewhat fewer gates.
Who would have thought that this makes such a big difference.

FaberUnserzeit, 2021-09-22:
@hneemann After having understood the CLA-logic, I expect a big difference. I'm currently in the process of improving the divider. In spite of the huge number of gates a 4 bit CLA adder performed slightly better than a 4 bit ripple carry adder in minetest/mesecons. So I started, based on your CLA-adder example, to make bigger CLA-adders with Digital and designed comperators in all sizes between 1 and 16 bits, so I don't need more gates than necessary for each stage and I don't have to cascade more than two of them to reach 32 bit. 
[add_comp.zip](https://github.com/hneemann/Digital/files/7206887/add_comp.zip)
The (realtive to the current state) higher bits of the divisor can be OR'ed together and go into the comparison as another bit, which saves comperator bits. I could go up to 32 bits, as well, it's mainly copy and paste and entering numbers for every splitter/merger, almost meditative, but while the number of gates rises linearly with the number of bits, the number of gate-inputs rises dramatically. I don't know how big these things get in silicon, how many inputs one output can drive, and so on. [add_comp.zip](https://github.com/hneemann/Digital/files/7206887/add_comp.zip)
My interest lies in minetest/mesecons, because minetest is an interest of my daughter (12), that we can share, and in this subject I see a potential to awaken interest for digital logic and what can be done with it. 
For implementations in mesecons that use gates with a large number of inputs, the logic has to be inverted in parts, because there are only gates with two inputs (or three, if you program, what is called "FPGA" in mesecons with simple logic), but there is no difference between zero and high-Z, so big OR- or NAND gates can be made by wire-OR'ing everything with diodes or inverters. So instead of cascading two-input gates in a tree-manner to keep the number of propagation steps low, a better result can be accomplished this way. My goal in this context is to produce a video series together with my daughter that introduces mesecons and shows the way from demonstrating what gates do up to designing a simple computer somewhere in between Ben Eater's and RJ45 Creates' creations just using gates and "FPGA"s and a delayer for clock-generation. I have come quite far relative to my low standard with something similar to Ben Eater's design. I have 64 bytes of RAM (instead of 16), thus I have a 6 bit PC and variable instruction size, because I need byte-size parameters in some instructions. I have 3 bits for the microstep-counter, which is reset by microcode at the end of the instruction. There are two ROMs with 6 bit addresses for the microcode that I can program by placing AND-Gates. There is also logic that switches between them to extend the number of possible opcodes. One-byte instructions lay in the first ROM and two-byte-instructions are i
[add_comp.zip](https://github.com/hneemann/Digital/files/7206887/add_comp.zip)
ntroduced by a certain value of the opcode-bits, that increments PC and loads the parameter, and uses the lower 4 bits of the instruction byte as part of the address for the second ROM, selects that and resets the microsteps-counter. This works fine, I placed an adder over the PC (a modified 74163), implemented a relative jump in the first ROM, an absolute jump in the second ROM, and as I wanted to integrate the ALU with multiplier and divider (and thus 4 registers including C and D for remainder of division or upper byte of multiplication) and connect the control lines to my ROM, i figured out, that I have to choose: I can address the registers and ALU-parts directly (like Ben) and bind some (or most) opcodes to certain registers (like 8086's MUL), or I can enumerate the registers and address them with parts of the instruction byte (like RJ45 Creates, the AVRs or most modern CPUs). Instead of connecting the ALU-control lines to the ROMS and wasting a whole slot in my first ROM for a simple register ADD with fixed registers, I started looking for inspiration on youtube, found RJ45 Creates. This is how I've come here. :-D
[image]

hneemann, 2021-09-22:
@FaberUnserzeit 
What a great project! Even more so if it is also fun for the kids. :smile:

When NAND or NOR gates are built in silicon, as many p-channel (NOR) or n-channel (NAND) MOS-FETS are connected in series as the gate has inputs.Since a FET has a residual resistance, the gate can drive following gates less and less well as the number of inputs increases.  The maximum fanout therefore decreases more and more, or the circuit becomes slower and slower, because the recharging of the subsequent gates takes longer. Since p-channel FETS are more affected by this, since the residual resistance is higher than for n-channel FETs, NAND gates are easier to build.
I remember reading once that gates with more than 5 inputs are rarely used. If more inputs are needed, the gates are cascaded, but this also increases the runtime. How it is actually implemented will depend on the individual case.

commit b47c2dfc (2021-09-16): allows to set the oscillationDetectionCounter in the circuit specific settings; closes #837