hneemann/Digital #838: RAM losing data when starting simulation

SgtSchultz71, 2021-09-17:
I've noticed that when I load my RAM with data before starting a simulation, _sometimes_ that data disappears. I have to then go back into the RAM module and type in the data once more. What is also strange is that this only seems to happen in the first memory cell. I am not sure if this is a bug or somehow intentional but I was curious as to why this occurs.

[image]

hneemann, 2021-09-17:
This seems to be a problem similar to #835

SgtSchultz71, 2021-09-17:
Yes indeed! I failed to find that particular entry to the forum but it makes sense. Thank you!

closed 2021-09-17 by SgtSchultz71

hneemann, 2021-09-17:
@SgtSchultz71 
You're welcome! :smile: