hneemann/Digital #841: no bi-directional pins?

mitolegy, 2021-09-23:
hi
the settings on Input pins called "is three state input" and "no zero output" that allow input pins to become bi-directional IO pins don't seem to be working. I've used them before this way so I know it worked in the past. but now when I have just two 1-bit inputs hooked directly into each other with the settings enabled on both I get a short circuit error as soon as one changes state. is this a bug? or am in just doing something wrong?

hneemann, 2021-09-23:
Each pin is always bi-directional. There are no pins that are not.

The settings you mention only affect how the pin can be controlled by the mouse. With the option "is three state input" mouse clicks will not switch between the states "High" and "Low", but mouse clicks will cycle through the states "Low", "High" and "High-Z". If additionally the option "No Zero Output" is selected, the output changes between "High" and "High-Z".

And as with all other signals, a short circuit is detected when a High signal is directly connected to a Low signal.

mitolegy, 2021-09-23:
Interesting.
I did some further testing based off what you said. it seems that when input components are initialized for the simulation. they're understandably initialized with their default value, which by default is zero. even if zero or "low" is disabled. they still start out with "low" as their value. all you have to do is manually switch them once and they start behaving as they should however. 
if you like me for example have two wired right into each. they'll cause a short. as one switches to a high, directly into the low value of the un-updated one.

hneemann, 2021-09-24:
> it only seemed to fix it for 1-bit components. multi bit components still want to fight. even if the default value is set so all pins would be high.

I am not able to reproduce this. Could you please attach an example circuit that shows this error at startup?

mitolegy, 2021-09-25:
> > it only seemed to fix it for 1-bit components. multi bit components still want to fight. even if the default value is set so all pins would be high.
> 
> I am not able to reproduce this. Could you please attach an example circuit that shows this error at startup?

sure.
here's what I'm getting. 
in this picture the simulation is running. 
[image]
as you can see. the input pin is low. right? no, it is but its not supposed to be. the low state is disabled on this input component. however its default starting value is zero.

as you can see here. toggling the input component fixes the problem. the lowest one is in the Z state. after having been toggled on and off once. 
setting the default value of the input component to 1 accomplishes the same thing. simply skipping step 1 where the component is in the low state.
[image]

furthermore. more investigating seems to show that multi bit inputs never enter the Z state. in the image below. the lowest pin, like in the single bit example, has been toggled on and off. however here it did not enter the Z state. on the 3-bit input, the zero state is disabled.
[image]

here is the Digital version I'm using.
[image]
the circuit file is attached
[Input component bug.zip](https://github.com/hneemann/Digital/files/7229479/Input.component.bug.zip)

hneemann, 2021-09-25:
There must be a misunderstanding. I think I should clarify the help text for this setting.

This setting only affect how the component reacts to mouse clicks. They do not affect the default value set for the component, nor how multibit inputs react. Multibit inputs do not react diriectly to mouse clicks, because a dialog opens in which you can enter whatever you want.
The background of this setting are special circuits, which are built with relays, where only high and high-z are used in the logic. So that these can be controlled with the mouse, the low value can be skipped with the mouse operation. In these circuits a low level leads to a short circuit. If the low level could not be skipped in such a circuit with a single bit input, the circuit would not be operable.

See also #177

mitolegy, 2021-09-25:
so how am I supposed to do bi-directional IO then? especially for multi-bit components where I'm currently struggling. 
right now if I feed a signal into a multi-bit input component. I get a short. with or without the setting enabled. where as before I swear they would start in Z. and then let me feed a high into them without complaining. 

infact for reference I pulled up an earlier circuit I made a while back. and as you can see. when running as a component. the output on the 8-bit data bus (D0-7 on components "PIP-AX08 CPU rev 2" and "AX08 Memory unit (rev2 )") is Z. however when inside the actual circuit its 0. and if I try making the circuit output data to the port i get a short circuit error.
[image]
inside PIP-AX08 CPU rev 2
[image]
if this is not how to do bi-directional ports then id very much appreciate an example of how to do them

hneemann, 2021-09-25:
Let me explain how inputs and outputs work in the simulator:

When a circuit is embedded in another, all inputs and outputs are replaced by a direct connection with a wire. Inputs and outputs exist only in the top-level circuit. They are only GUI components that allow setting values or displaying them.
You can easily check this: If you set the wrong bit number at an input in a circuit, the circuit will not start. The error message appears that the bit number does not match. If you embed this circuit, however, it works without problems, because the input is removed and replaced by a direct connection. The bit number that was set does not matter anymore. A pin in the simulator is exactly what a pin is in reality: an electrical contact without its own function. The pin itself has no properties, only the circuit that works inside has.

When a circuit is embedded a second time, a second copy of the circuit is created and inserted at the new location by creating the appropriate direct connections. It is like a C program where each function has an "inline" attribute set.

See also #438

hneemann, 2021-09-25:
@mitolegy 
I have thought again where the misunderstanding could lie: 
Do you know that an input can also have a default value "High-Z"?

closed 2021-10-02 by hneemann