hneemann/Digital #842: Random short on a data bus (simple 8 bit processor)

lefuturiste, 2021-09-23:
Hi.

I built a simple 8 bit computer to do Fibonacci sequence.

All the assembly code and files is here :

https://github.com/lefuturiste/8bit-computer

but it errors randomly (sometimes the computer run fine for 5 seconds other time it crashed immediately) :

```
Error calculating a step
More than one output is active on a wire, causing a short circuit.
```
It's highlighting the data bus.

But for me, I don't know how something like that could happen with the data bus, may be is it because of the registers, I don't know.

I also tried with very simple program that load things into the registers and it didn't work either;

Can someone can help be to have way to debug this issue?

hneemann, 2021-09-24:
A pull down is missing, which leads to certain wires being undefined at times, which can lead to unpredictable effects.

[image]

So many wired Or's are used that it becomes confusing. Maybe these should be replaced by real OR gates. In my opinion it is not a good idea if a design relies so massively on High-Z states and pull resistors have to be inserted everywhere.

lefuturiste, 2021-09-24:
Ok, many thanks for pointing the missing pull down it worked perfectly. 
Now I want to know more about how I can avoid pull resistors.
What are "wired Or's" and how these can be replaced by OR gates.

For example to say to a register to a register to store smth I use a "STORE" flag line which is pull down. What's the better solution to your opinion ?

[image]

hneemann, 2021-09-24:
Replace this:
[image]
by this:
[image]
or this:
[image]

closed 2021-09-27 by hneemann