hneemann/Digital #843: Toggle real-time-clock

PaulKappmeyer, 2021-09-25:
Hello,
I’m wondering if it could be possible to add a keyboard shortcut for toggling the real-time-clock while the simulation is running.
Since there is already a shortcut for single-stepping the clock (C Key), it would be very helpful to be able to stop the clock in the simulation at some point and single-step the program.

hneemann, 2021-09-25:
Since the synchronization is different in the real-time clock mode and in the single-step clock mode, this is difficult to implement. 
But you can use the break component. This way you can define a condition, automatically cycle the circuit until the condition is met and then continue in single step mode. It almost feels like a breakpoint in a programming language. In my opinion debugging is also easier, because it always stops at a defined point.

PaulKappmeyer, 2021-09-28:
So I’m trying to build Ben Eater’s 8-Bit Pc with Digital, which is why I like the idea of changing the clock-mode while simulation.
But at the moment I’m using this circuit, which seems like a good compromise.
[image]

Thanks for your answer!

closed 2021-09-28 by PaulKappmeyer