hneemann/Digital #844: XOR symbol in text

eliask05, 2021-09-25:
This question is probably OS-specific. For reference, I am using Digital under windows.

[image]

I like to include equations in the circuits when I give problems and solutions to students. Using the windows character map application, I can insert summation, product and XNOR symbols since they are included in the default font (Arial). The XOR symbol, however, is not included. In the attached screenshot, edited in paint,  I got it from the symbol font but this is not possible in Digital since the text is in a single font and that is Arial.

Any suggestions on how I can get the XOR symbol, if at all possible?

hneemann, 2021-09-26:
It must be a OS specific problem. On my system (Debian) it works without an issue:

[image]

But I am not a Windows expert. In fact, I'm not even a Windows user, so I can't give advice.

Here is the circuit i have used to create the screenshot above, just in case it is a unicode codepoint problem:

[xor.dig.zip](https://github.com/hneemann/Digital/files/7230869/xor.dig.zip)

eliask05, 2021-09-26:
Thank you! It actually displays fine on my Windows system, but how did you enter the symbol in the text box?

eliask05, 2021-09-26:
OK, I figured it out. I used BabelMap which is much more powerful and user friendly than character map.
Thanks for the help, I am closing the issue.

closed 2021-09-26 by eliask05

hneemann, 2021-09-26:
@eliask05 
Up to now the LaTeX commands `\sum, \prod, \wedge, \vee` and `\neg` are available.
I will add `\oplus` and `\odot`.

Is something else missing?

eliask05, 2021-09-26:
Great! I wasn't aware of the LaTeX commands. I think the following may be also useful:
\pm \mp \div \cdot \times and \otimes

hneemann, 2021-09-26:
@eliask05 
I have added them. Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

eliask05, 2021-09-26:
Thank you! They work great. I have modified the "TextFormatExample.dig" example file in case you wanted to document them
[TextFormatExample.zip](https://github.com/hneemann/Digital/files/7232258/TextFormatExample.zip)
.

hneemann, 2021-09-27:
@eliask05 
Thanks! :smile: