hneemann/Digital #862: Request: inverter with bubble at the input

eliask05, 2021-10-16:
Is it possible to have the choice on the inverter to invert the input, rather than the output?

Thanks!

hneemann, 2021-10-17:
Does that make a difference?

hneemann, 2021-10-17:
There are already four shapes for the inverter. To be honest, I don't think it makes sense to add more.

eliask05, 2021-10-18:
I understand. No problem but I could use some help:
 I want to create my own shape but I am afraid I don't follow the instructions in the manual.
I created a simple  inverter circuit (called invn.dig, see image below). I now want to attach my svg file (included in the zip file) to this circuit. How do I go about it? The manual states: "If the file was saved, it can be imported with Digital."
I don't see an import menu anywhere. How do I proceed?
Thanks!
![invn](https://user-images.githubusercontent.com/41453704/137650562-b02e4cc9-2685-4c1f-b70a-85295ce36e37.png
[invn.zip](https://github.com/hneemann/Digital/files/7361138/invn.zip)
)

hneemann, 2021-10-18:
In the circuit specific settings you can select the shape type "User Defined". Then you can export a SVG template. This file can be edited and then imported again. All elements can be changed or deleted, only the red and blue dots must be kept, so that the contacts can be identified during the import.

The result looks like this: [invn.dig.zip](https://github.com/hneemann/Digital/files/7362134/invn.dig.zip)

eliask05, 2021-10-18:
Thank you! I am closing the issue.

closed 2021-10-18 by eliask05