hneemann/Digital #877: add a new counter 74ls90 in Digital

B-kai-young-N, 2021-11-08:
This is one of our assignments, our teacher ask us to create a new chip which Digital does not have yet, and we choose 74ls90, asynchronous two - five - decimal addition counter.
We want to add it into the lib of Digital, your reply will be greatly appreciate.
[7490.zip](https://github.com/hneemann/Digital/files/7496551/7490.zip)

hneemann, 2021-11-11:
There are no test cases and the power supply (see other chips) is missing.

B-kai-young-N, 2021-11-13:
> There are no test cases and the power supply (see other chips) is missing. Shouldn't these be added?

Thank you for your reply! We are very sorry to have wasted your time due to our lack of consideration.
This time we have added a test model to the original circuit to verify its basic functionality and made a PDF file to describe the function of this chip.
[7490andTest.zip](https://github.com/hneemann/Digital/files/7532164/7490andTest.zip)

hneemann, 2021-11-14:
No need to apologize! I'm happy to help! :smile: 
I don't think that a description as a PDF is necessary, because a data sheet is freely available.
However, the test case does not contain a clock signal. So it is not checked if the counter counts correctly and also overflows correctly.

B-kai-young-N, 2021-11-14:
Nice of you!
But I think I don't fully understand what you mean, because for this chip,  if we want the whole chip to work in a specific counting mode, it is necessary to make some circuit connections on the outside of the chip, and inside the chip, I can only check its basic counting mode.
I have already added another counting test case inside the chip.
By the way, for insurance purposes, I let clock(CKA and CKB) signal "0-1-0-1-0……" instead of "c".
[7490(3).zip](https://github.com/hneemann/Digital/files/7533687/7490.3.zip)

B-kai-young-N, 2021-11-14:
> No need to apologize! I'm happy to help! 😄 I don't think that a description as a PDF is necessary, because a data sheet is freely available. However, the test case does not contain a clock signal. So it is not checked if the counter counts correctly and also overflows correctly.

Nice of you!
But I think I don't fully understand what you mean, because for this chip, if we want the whole chip to work in a specific counting mode, it is necessary to make some circuit connections on the outside of the chip, and inside the chip, I can only check its basic counting mode.
I have already added another counting test case inside the chip.
By the way, for insurance purposes, I let clock(CKA and CKB) signal "0-1-0-1-0……" instead of "c".
[7490(3).zip](https://github.com/hneemann/Digital/files/7533738/7490.3.zip)

hneemann, 2021-11-17:
> But I think I don't fully understand what you mean, because for this chip, if we want the whole chip to work in a specific counting mode, it is necessary to make some circuit connections on the outside of the chip, and inside the chip, I can only check its basic counting mode.

If for a special counting mode an output needs to be connected to an input outside of the chip you can write a test case and set this input in a way it matches the expected output value. This way also the other counting modes can be tested.

B-kai-young-N, 2021-11-18:
> If for a special counting mode an output needs to be connected to an input outside of the chip you can write a test case and set this input in a way it matches the expected output value. This way also the other counting modes can be tested.

After your enlightenment, I finally realized that in test case, if a specific working mode needs to connect A to B, it is not necessary to actually connect in the circuit, just assign A value to B is ok.
Then I added several test cases in the circuit for every specific working mode and passed.
[7490(4).zip](https://github.com/hneemann/Digital/files/7560106/7490.4.zip)

commit 0d9ae65a (2021-11-18): adds the 7490 counter, closes #877
closed 2021-11-18 by hneemann