hneemann/Digital #879: ROM content issues when moving from Windows to Linux

coreidcc, 2021-11-19:
How to reproduce:

On Windows
1. place an EEPROM (-> change #no address bits to 4)
2. edit content of rom - set[0] = 1, set[15] = 1 (leave everything in between == 0)
3. save test.dig
On Linux 
1. try to open test.dig => error

cause-exception : java.lang.RuntimeException cause-message : de.neemann.digital.core.Bits$NumberFormatException: The string 14*0 is not a valid number (pos 3)!

content of test.dig
<?xml version="1.0" encoding="utf-8"?>
<circuit>
  <version>1</version>
  <attributes/>
  <visualElements>
    <visualElement>
      <elementName>EEPROM</elementName>
      <elementAttributes>
        <entry>
          <string>AddrBits</string>
          <int>4</int>
        </entry>
        <entry>
          <string>Data</string>
          <data>1,14*0,1,239*0,12</data>
        </entry>
      </elementAttributes>
      <pos x="340" y="260"/>
    </visualElement>
  </visualElements>
  <wires/>
  <measurementOrdering/>
</circuit>

hneemann, 2021-11-19:
Could this be a duplicate of #826?

coreidcc, 2021-11-19:
My bad. You're absolutely right. my bad I made a stupid mistake when updating (old Digital.zip) was still in my download folder thus a Digital(1).zip was created. thus I replaced 0.27 with 0.27 :(

closed 2021-11-19 by coreidcc