hneemann/Digital #882: First impressions on the Digital simulator

Codified-Vexing, 2021-12-02:
I've tried Digital looking for a more modern version of Logisim and that's how I ended up here. This is my feedback after trying my usual workflow for a short time. I hope you don't find this post too long and boring.
### Good
**First here's some good things. Digital has some good ideas I've always missed in Logisim:**

The infinite canvas is a no-brainer that Logisim couldn't handle. No more do I need to move the entire circuit when I need some more room over the top or the left side of the circuit!
Right-click dragging to pan the view is a very refreshing experience compared to Logisim's scrolling. I just wish the control was the same while running the simulation, for consistency.
The component search on tree view is something I didn't know I needed until I saw it!
In the component options/properties you can actually type numbers instead of having to pick them from a drop-down. This is excellent and I don't know why Logisim wouldn't allow me to do that.
Having a component description when hovering with the cursor is rather handy, but I think it's enough to have that feature on the component tree view. Verbose tooltips randomly appearing on the canvas is annoying and unnecessary, given how easy it is to find the component on the tree view.
Being able to automatically rename all the tunnels of the same net is great!
The dark colour scheme is nice, but you could have the menus also have a dark mode.
Having the option to choose which inputs are negative logic is delicious.
Being able to load the contents of a ROM from a file automatically it's quite handy. Too bad you can't do the same for RAMs. To be honest I feel there should be just a single generic "Memory" component where you set available inputs and outputs in the properties.
Automatically generating gate and flip-flop networks, and Karnaugh Maps from truth tables is a cool feature, although I don't see a use for it myself. Maybe for learners or students is great.
There's a LOT more components. Let me list some I find particularly interesting or useful.

- A push switch with an integrated LED. It's that simple. Let me also have a latching switch version please.
- The mechanical devices like rotary encoders, stepper motors and relays are odd, but I might fancy them.
- The "monoflop", which is a monostable multivibrator is something I'm grateful for. My brain often comes up with solutions involving that kinda of logic.
- Finally a RAM with true dual-ports. The lack of such component was really bothering about Logisim. Quad-port RAM when?
- The library of 74xx components is pretty comprehensive. Rather handy if you need a bit extra authenticity in your simulation when you're planning on building the circuit in real life.

### Bad
**Now the worse things. There's some things in this list that could be construed as suggestions for features or enhancements. Tell me in the replies if I should post an issue report for each of these:**

The GUI is definitely better and more streamlined than Logisim, but the user experience is worst somehow.
I really miss having a side panel with tabs on different stats and options of the simulation.
The components panel feels schizophrenic with the symbol representations only showing up once the component is placed and being all sizes and shapes. There should be a minimalistic icon for each component, all icons the same size.
The side panel could also have component options/properties tab. So it automatically displays the options for a selected component without having to handle different pop-up windows all the time.
The zooming and simulation/stepping controls take too much space in the toolbar. I'd rather have the toolbar hold a set of favourite components.
There's no way to change the clock speed during a running simulation. This is a minor issue, but I have a pet peeve for this.
Digital seems rather picky with user mistakes when running the simulation. Couldn't floating/unconnected pins should be assumed to be in Hi-Z state?
Why can wires be drawn at weird angles? Anything beyond right angles and 45deg angles is a nuisance.
Wire splitters being unidirectional is garbage. The "bus splitter" as a separate component doesn't even work as intended either.
The DIP switches can't be grouped. Why do I have to handle individual switches?
I think it's weird to have two versions of Flip-Flops "regular" and asynchronous. I'd rather name them as "D Flip-Flop" (synchronous) and "D Latch" (Asynchronous), for example. But ideally, Flip-Flops should have an option to make their inputs asynchronous and therefor reduce the component spam.
This simulator has a bad issue with many separate components which are only a slight variation of each other. There should be generic models of a certain component with options to set up various behaviours and available pins.

### Ugly
**The good things can make me overlook the bad things if they are reasonable improvements over what Logisim offers. But these things make me hesitate:**

The symbol representations are a mess. The pins seem to be placed tightly to make the rectangle the smallest possible, but with no distinction or alignment on related pins. Makes it hard to understand at a quick glance.
The wire handling is bad. It's too easy to draw lines accidentally when you just intend to select a device or box-select something. Also, I can't select particular segments of wires properly. The only way to select wires is to box-select them and it will select segments even when not fully enveloped by the box. It's impossible to select small parts of wires that way.
Being able to start a wire from anywhere on the canvas only makes it more annoying. Logisim only allows drawing the wires from component pins. That's much better.
In Logisim you can set data in inputs, registers, flip-flops or memories while the simulation is running. You even may see the contents of those components inside the symbol. I wish Digital worked like that too, instead of having to open a pop-up window and click "apply" for any change. This is miserable. Being able to type in values or use the scroll wheel on them is a much better experience than the ugly static symbols.
I think the Hex Editor for editing ROMs and RAMs is a very important feature in a logic simulator. This goes along with the point on being able to see a memory's contents on the symbol because it makes these components to be more recognizable and distinct from other components.

### What Else
**There's a couple of things I think would improve Digital if it had them, but regardless isn't a deal-breaker:**

Searching for tunnels and labels. it'd be great if I could type words and the canvas automatically pan to occurrences of that word in component labels or tunnel net names. Much like searching words in a PDF schematic of a circuit.
A tab of the side panel that holds state/values of selected inputs and outputs. That way I don't have pan all over the circuit too see multiple things happening at the same time. Similar to the "Registers" tab in Logisim, but more versatile.
A tab of the side panel to view the hierarchy of subcircuits and also open them in a new window when double clicked. Even better if there's support for tabs on the canvas, so you switch between several circuits in the same Digital window.

I hope you take pride in developing this software. It has great potential. Thank you for your attention.

MPagliari, 2021-12-28:
Why I liked Logisim over Digital: To search "the Google" and find data and information on Logisim is easy as this is not a regular word like Digital. I find myself trying to filter out every search result with the word Neemann appended to help filter out this issue.
While the word Digital is fairly descriptive to the application it does cause ambiguous search results. I would like to see the name appended with another name expression to help filter search terms. This would result in easier searching for new users and easier discovery of addon components and libraries. 

While its a huge undertaking to rename an already set in motion project I do believe this naming convention would help the Simulator gain more ground over the older Logisim, this does seam to be more of a social issue as it would take trying to convince the community to use a naming expression when writing articles about Digital.

To round off I do like Digital over Logisim from a expandability point of view and do agree with why we should switch from it. I just have issue with the name sorry about that.

Great program though 

-Peace

jxu, 2024-01-12:
@MPagliari #151