hneemann/Digital #886: Export visualisation of instantiated generic component: Feature Request

adiehl96, 2021-12-10:
I am currently writing a report where I would like to showcase the functionality of a generic component. However, I only seem to be able to export an svg file from a generic but uninitialised component. This means that in order to show this component in its instantiated state (with some values set and some programmed wires and logic appearing), I have to take a screenshot.

Would it be feasible to create a functionality that when the user clicks on export svg, while the component is being simulated, that the currently displayed circuit is exported?

hneemann, 2021-12-14:
Add a "init" component to the circuit.
With this component you can parameterize the circuit and also create a concrete circuit, which can then be exported.

adiehl96, 2021-12-15:
I just figured out the issue and here's how to reproduce it:

Create a generic component.
Create Init with default value.
Click the play button that starts the simulation of the circuit.
Click on File -> Export -> Export PNG Large.
Open the png.
See that only the uninstantiated component is exported.

However, the workflow to export the instantiated component is the following:

Create a generic component.
Create Init with default value.
Right click on the init box.
Click on Create in the init box dialogue.
Click on File -> Export -> Export PNG Large.
Open the png.
See that only the instantiated component is exported.

Not sure if many people have trouble finding this functionality, but I sure did! I would have expected that "what you see is what you get" thus that exporting during simulation would produce the second behaviour rather than the first one.

commit 6a4dbbd6 (2021-12-15): graphic export also exports running generic circuits correctly; closes #886
closed 2021-12-15 by hneemann