hneemann/Digital #890: Add Global Reset to Graphic RAM

TheWildJarvi, 2022-01-02:
Hello, I am working on a WIREFRAME 3D renderer and I am faced with the painstakingly long task of clearing the display before rendering the next set of lines. I am using the double buffer feature of the display and it would be very convenient if you could have a global reset( I have no preference for async or sync) for each buffer.

Ideally the color seen at the data input will be written to all addresses (of a single buffer of the display) at once. This would enable me to render frames much faster as each line may only be several hundred pixels long, and for a cube of 12 lines thats 1200 pixels(really quick) vs rendering a whole frame of 1024x1024 is over 1 million pixels I need to iterate through, just to clear the screen. I hope this is persuasive enough. I have been using Digital for a long time now and I really appreciate all the work you have put into it.
Thanks a lot
 
TWJ

hneemann, 2022-01-03:
In this case it would be better to implement something similar to the Amiga Blitter chip. This chip could move and erase areas in graphics memory and was even able to draw lines completely in hardware.

TheWildJarvi, 2022-01-04:
Hmm, that seems like a little more feature-full than I really need. But any update to the graphic ram would be very much appreciated.

hneemann, 2022-01-05:
While this would offer more features, it would be easier to implement because it would not require direct access to the pixel data. One could increase the resolution almost arbitrarily.
One would only have to design a suitable interface, which is easy to access.
On the other hand, this would be more something for a special plugin.