hneemann/Digital #891: Document test error

VitalyAnkh, 2022-01-04:
I got test error when running `mvn install` with the most recent source code (commit hash is 1951d7aee84ae756e42f72c05faae7d19ad681ac).
```
[ERROR] Failures: 
[ERROR]   DocuTest.testDocu:337->startFOP:287
[INFO] 
[ERROR] Tests run: 815, Failures: 1, Errors: 0, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  01:35 min
[INFO] Finished at: 2022-01-05T04:42:43+08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:3.0.0-M1:test (default-test) on project digital: There are test failures.
[ERROR] 
[ERROR] Please refer to /home/vitalyr/.cache/paru/clone/digital-git/src/digital-git/target/surefire-reports for the individual test results.
[ERROR] Please refer to dump files (if any exist) [date].dump, [date]-jvmRun[N].dump and [date].dumpstream.
[ERROR] -> [Help 1]
[ERROR] 
```
The corresponding report for this individual test is copied from `digital-git/src/digital-git/target/surefire-reports`: [TEST-de.neemann.digital.docu.DocuTest.xml](https://gist.github.com/VitalyAnkh/e9584f6d50010a05dc7f97587e56eee9).
Thanks for making this great software!

hneemann, 2022-01-05:
This seems to be a problem with Java 17 and jacoco.
I'm going to take a look at that. 
In the meantime, you can remove the jacoco plugin that is only used to obtain the test coverage.

jmb6, 2022-06-22:
I am having the same issue as OP, but replacing jacoco 0.8.4 with 0.8.8 in pom.xml did not fix it. I still get the same build error. I'm using java 18 but I get the same build error on java 11 and java 8 as well.