hneemann/Digital #892: Unable to run Analysis with flip-flop

reifsnyderb, 2022-01-15:
When creating a logic circuit, I am unable to run an Analysis with either a flip flop or NOR gates set up to function as a flip flop.  Without running an analysis I am unable to create the files to flash an ATF1504.

reifsnyderb, 2022-01-15:
File attached....

[600 XLM R08 MMU.zip](https://github.com/hneemann/Digital/files/7874427/600.XLM.R08.MMU.zip)

hneemann, 2022-01-15:
The analysis algorithm is not able to recognize a flip-flop constructed from simple gates and to replace it with a corresponding one. Therefore, the flip-flops that are available as built-in components must be used. The built-in D, T and JK flip-flops can be used. RS-flip-flops cannot be used due to the inherent instability.

reifsnyderb, 2022-01-15:
Got it working.  Thank you!

closed 2022-01-15 by hneemann