hneemann/Digital #913: What does this error mean and how do I resolve it?

reifsnyderb, 2022-02-06:
Hello,

I am getting this error when running an analysis:

Error during simplification
Too many variables used in Z^{n+1},
12 are allowed variables but 19 were found.

I haven't found any information on this issue and don't know how to resolve it.  

I've attached the file.  I am trying to use this to upload to an ATF1504 CPLD.

[CPLD_RAM_2MB.zip](https://github.com/hneemann/Digital/files/8011295/CPLD_RAM_2MB.zip)

Thanks!

hneemann, 2022-02-07:
The simulator uses QMC (Quine and McCluskey) to simplify logic equations, and the problem with QMC is that while it always provides an exact minimum solution, the memory consumption and computing time increases exponentially with the number of input variables. Therefore QMC is not suitable for functions with many inputs. To avoid that it takes too long until the result is available, the calculation is not performed if there are more than 12 inputs.
With such a large number of inputs you would have to use e.g. the heuristic Espresso algorithm, which is faster and uses less memory, but does not always provide the minimum solution.
Unfortunately, I have not yet got around to implementing the Espresso algorithm.

see also #495

reifsnyderb, 2022-02-07:
Is there a way to increase the number of inputs at your own risk?

hneemann, 2022-02-07:
Yes! You can start the simulator with the command line argument `experimental`.
This avoids the error message. But in this case you should also increase the allowed max memory consumption.
```
java -Xmx8192m -jar Digital.jar experimental
```

reifsnyderb, 2022-02-07:
I terminated the program after about 4 hours.  Memory consumption was up over 10 GB.  I really like the program but will have to use it for logic simulations and smaller CPLD's only.

hneemann, 2022-02-08:
As I said before: Exact solution algorithms are of order O(2^n). These are not suitable for minimizing equations with more than 12-14 variables. And this is sufficient for the application for which the simulator is intended (teaching).
But without a heuristic, non-exact solution method like e.g. the Espresso algorithm, it is not possible to find a solution in many cases.

hneemann, 2022-02-11:
see #918

closed 2022-02-11 by hneemann