hneemann/Digital #919: wires freak out and displace after taking a step back

The-Devil-Itself, 2022-02-12:
wires rearrange themself when ctrl-z has been used,
it only happens if there are more than 80 components present

hneemann, 2022-02-12:
Are you able to reproduce this issue? It seems to be a duplicate of #364.
I can not reproduce this error despite all attempts. Therefore, I can not fix it.
If you know a way to reproduce this problem I could finally fix it.

The-Devil-Itself, 2022-02-12:
you can reproduce it by making an 80+ component cicruit with wires, the move all of it from the original place to anywhere else,
then ctrl z, and you'll see the displaced wires, i'm able to reproduce the issue

hneemann, 2022-02-12:
Just tested: Open examples/sequential/conway/Conway.dig (1000+ components and wires), move any part or the compete circuit arround, press Ctrl-Z and no problem whatsoever.

The-Devil-Itself, 2022-02-12:
i was unable to reproduce it with any exmple,but there's a working instance of the bug

[tinyest computer.dig.tar.gz](https://github.com/hneemann/Digital/files/8053301/tinyest.computer.dig.tar.gz)

hneemann, 2022-02-12:
When I open this circuit, and move parts of it or all of it, ctrl-z works without problems.

Are there any files named like `Digital_2021-09-22_16-16-12.log` in your home-folder? It is the same folder the file `.digital.cfg` is located.

hneemann, 2022-02-12:
The simulator uses event-sourcing for the undo. So all changes since the last loading of the circuit are stored and if a step is to be undone, the original circuit is restored and all changes except the last one are executed again.
It seems that it is somehow possible to make changes bypassing this system. If then an undo is executed, chaotic conditions arise. And I have no idea how a change can be made bypassing the event-sourcing system.

The-Devil-Itself, 2022-02-12:
where can i find this log in the linux version? bc i don't see any in the folder . or in an extended search through the file system

hneemann, 2022-02-12:
On a linux system such a file is stored directly in the home folder.
If there is none, none was created. Too bad!

The-Devil-Itself, 2022-02-12:
okay, i found some clue .
it never happens, if the file has been saved.
https://user-images.githubusercontent.com/99536568/153712370-eb5b8b91-6571-445f-8b63-561407c34904.mp4

The-Devil-Itself, 2022-02-12:
And it only happend to the oddly oriented wires, not any of the horizontal or vertical

hneemann, 2022-02-12:
Thank you for the video! I have watched it many times and even more I have tried it with this circuit and here it works perfectly every time.

xotmatrix, 2022-02-12:
I've experienced rare problems exactly like this going years back. I never figured out how to trigger it reliably but I do specifically recall it happening with diagonal wires.

The-Devil-Itself, 2022-02-14:
Yes, i will happily test it
and i am able to replicate it reliably whenever i want to.

hneemann, 2022-02-15:
Could you please try v0.29-5-g86af23b and v0.29-3-g32f924c you can find [here](https://infdigital.dhbw-mosbach.de/)?

I am not sure if any of the builds will fix the problem. Since I can't reproduce the bug, I essentially guessed. :-(

The-Devil-Itself, 2022-02-15:
it fixed it, i cant produce the problem in this build

hneemann, 2022-02-15:
Which one does fix it?

The-Devil-Itself, 2022-02-15:
i can't produce it in any of the 2,so i think the first