hneemann/Digital #930: Load ROM content from file

coreidcc, 2022-02-25:
During development I repeatably generate ROM content (4 in my case) by external tools (e.g. by assembler and micro-code generator). 

Each time I do need to do the following very repetitive work: 

a) Open dialog: Edit > Circuit Specific Settings 
b) Switch to Advanced Tab
c) Click Edit
d) For each ROM (i currently have 4)
    d1) double click on ROM
    d2) select File > Load 
    d3) double  click on Rom file (initially navigate to ROM directory)
    d4) clock OK
e) click OK
f) click OK

This is repetitive, error prone and consumes quite some time.

It would be awesome if the ROM content could be specified with a project-relative path and each time its content changes the ROM content gets reloaded. actually i would be already be a great improvement if such ROM content  would be loaded on simulation start. This would allow to just stop/start the simulation after a generation process.

hneemann, 2022-02-25:
It is possible to enable a autoload function. 
If enabled, a file is reloaded if the simulation starts.
You can find this option in the ROM settings.

coreidcc, 2022-02-25:
Thanks for the hint. I used the 28c64 (under Library>DIL Chips>EPROMs) that is why I didn't notice that option. I have choose to stay as close as possible to the actual hardware I am about to build. For the micro-code-roms this should be a straight replacement (CS, OE, WE pins are all hard wired). For the program ROM things look a bit different as CS, OE are memory mapped signals. There i would need to build a 20c64 part using the ROM chip.

Question: What do you thinks about the idea to implement all  Library>DIL Chips>EPROMs using the ROM chip you suggested to me? A the moment they are using the EPROM (not ROM). Or how about to add the same functionality the ROM provides to the EPROM?

hneemann, 2022-02-25:
There is also a command "program" to set the program memory. 
See the test for the "LD" command in _"examples/processor/ProcessorTest.dig"_.

hneemann, 2022-02-27:
Maybe the functionality of the generic circuits can be used for this task. I will think about it. 
Let me understand: You have three ROMs inserted in one embedded circuit each and to start the simulation you want to load all three with different data before starting the test or the simulation.

hneemann, 2022-03-09:
@coreidcc 
I decided to do without scripting and make all settings accessible via the GUI.
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/) build and let me know how it works.

coreidcc, 2022-03-12:
sorry for the late reply. i have multiple ROM/EEPROM in a sub-circuits (3 for micro-code in one sub-circuit, and another ROM/EEPROM one for a boot-rom in another sub-circuit). I do have 2 use cases
a) test in digital itself: in this case is like to write data into to the various ROM/EEPROM in order to test individual sub-circuits
b) i generate micro-code and/or boot-rom code with my micro-code generator / assembler. in this case i like to load newly generated rom-content on simulation start.

I will check out the pre-lease. Many Thx!

closed 2022-04-07 by hneemann