hneemann/Digital #934: Option to disable/configure remote port.

Parakleta, 2022-03-08:
Can we please have an option to disable and/or configure the remote port.  I use the program in a multi-user environment and the first student to run the program opens a port that anyone else can meddle with, and everyone else gets the warning "The remote port is in use! Is there an other instance running?"

Thanks.

hneemann, 2022-03-08:
Do you want to use the remote connection on multiple instances running on the same machine?
If not, the message can be ignored.

Parakleta, 2022-03-08:
I don't currently need to use the remote connection on multiple instances, but this change would be a convenient side effect.  The main issue is one of security.  You are opening port 41114 on every interface/address on the machine.  I have firewalls that close off access to these from external sources, but other users on the system are able to connect to opened ports on the local machine.

If you have an option/preference that allows an "ip:port" syntax with some invalid value (empty?) the this could be both controlled to allow multiple users, restricted to the loopback interface instead of opening a port on every interface, as well as being disabled.

hneemann, 2022-03-08:
Please try the latest [pre-release](https://infdigital.dhbw-mosbach.de/).

commit dcef287c (2022-03-08): The remote server is disabled by default; closes #934
closed 2022-03-08 by hneemann