hneemann/Digital #936: Verilog extern error

j054n, 2022-03-25:
I have installed IVerilog in Mac in:
/usr/local/bin/iverilog
with:
brew install iverilog
I have tested in terminal or from editor VSCode, included VCD viewing with GtkWave, without problems.
I have configured in your Digital simulator following steps in doc,
but I want to import simple Verilog modules from external code or external file gives error about not found iverilog
Now I have tested in a Windows environment and it goes fine.
Can you suggest a solution for Mac please?
Thanks in advance and regards,

Josan

hneemann, 2022-03-25:
In the settings dialog the path of the iverilog binary can be specified. 
Please check if the correct path is entered there.
If this is the case and it still does not work, I suspect a permissions problem.

closed 2022-05-08 by hneemann

j054n, 2022-10-11:
Thank you Mr. Neemann, the problem was with iverilog binary, it was a link in /usr/local/bin, it’s sufficient with searching the link destination in file system.
Regards,
Josan M