hneemann/Digital #940: Add optionto set memory, keyboard input, and terminal output via TCP interface

JAicewizard, 2022-04-01:
This would allow automated testing of code on a digital processor.

I am currently writing a compiler for a processor, and although there is an emulator IO doesn't 100% match the behaviour in digital.

It would be awesome if there was some way to run a circuit with a specific in and out file.
Adding a CLI interface would also be very nice, but adding it to the tcp interface doesnt require adding an interface.

hneemann, 2022-04-02:
There is a much simpler way to test processors. 
Please look at the circuit _examples/processor/ProcessorTest.dig_ to see how this is done with the example processor included in the distribution.

JAicewizard, 2022-04-02:
I dont want to test the CPU, I want to test code.
I know how to use digital tests, but I want to test the behavior of code generated by my compiler in digital.

hneemann, 2022-04-06:
There is already a simple TCP/IP interface to test programs in a simulated processor. A description of the functions can be found [here](https://github.com/hneemann/Assembler/blob/master/src/main/java/de/neemann/assembler/gui/RemoteInterface.java). Maybe this is already sufficient or could be extended.

hneemann, 2022-04-06:
However, it is possible to initialize the program memory to transfer a program to the processor..

JAicewizard, 2022-04-08:
I want a way to initialize this program memory, either by giving digital a path to a file, like when doing this manually, or giving it an array of bytes to use, whichever is more convenient for you. While I can do this manually, there is no way to do this automatically

hneemann, 2022-04-08:
It is possible to start the simulation via the remote interface, load a file into the program memory and run the program until a breakpoint is reached. And it is possible to read out all signals - in case of a processor all registers - via this interface.

closed 2022-05-08 by hneemann