hneemann/Digital #941: Digital doesnt recognise sub components, until a new subcomponent is made.

JAicewizard, 2022-04-01:
When opening my circuit digital puts "Shape XXX.dig is missing The file needs to be saved" (with a newline before "The").
Which is weird since all the circuits do exist. When I create a new subs-circuit, put in something simple, and then save the circuit, digital suddenly recognizes all the missing circuits.

This seems to point to a bug in the logic looking for components, as it does recognize the components after saving a new subcomponent.
Looking at https://github.com/hneemann/Digital/blob/f855750e5f09b95dc69b82073ed57355f430059f/src/main/java/de/neemann/digital/gui/Main.java#L1638, it looks like this is where it updates the setRootFilePath after saving the subcomponent.

I cannot figure out hoe to enable logs besides what is printed to the terminal.
Is there any way to find out what folder digital looks in for subcomponents?

hneemann, 2022-04-02:
The simulator searches for files to be included in the folder of the circuit that is currently open. If a new circuit is created with the "New" menu entry, and it is not yet saved, then there is no folder to search in. As soon as the file is saved, the corresponding folder can be searched, and the circuits contained there can be included.
If the menu item "New Sucircuit" is used, the current folder is kept and the circuits contained there can be used.

JAicewizard, 2022-04-02:
Maybe I should have given a bit more detail about my setup, an important detail is that I save the subcircuit in the. Same directory as the circuit that won't load custom subcomponents.

This is the file structure I am working with:
Computer.dig
components/cpu.dig
components/ram.dig
....

Opening Computer.dig gives the errors that cpu.dig etc cannot be found. 
Then if I make a new subcircuit(in a new window) and save it in the same directory as `Computer.dig`(so the root path should not be updated), the original window updates the custom components and the circuit works.
This noteworthy part is that its the same directory 

So the the simulator doesn't search for the files in the folder of the circuit that is currently open, if I interpret things correctly.

I also forgot to mention some of my setup,
(arch) Linux
Wayland
Java 18, also tested with 8 but same result.

JAicewizard, 2022-04-02:
Ok I have tracked this down a bit, it only happens when you open digital with a specific file using `java -jar digital.jar x.dig`. If you do this, this bug will happen, and also happen on consecutive openings of digital (without explicitly giving a file). Opening a curcuit via the "open" menu does work propperly.

So a reproducible setup:
X.dig
dir/Y.dig

X.dig uses Y.dig as a component
run `java -jar digital.jar X.dig` in a terminal.

and then you will get this error.

commit 13d6cd79 (2022-04-07): uses absolute file name if only the name is given via the command line; closes #941
closed 2022-04-07 by hneemann