hneemann/Digital #942: Reset Circuit for ICs

Ka3oose, 2022-04-05:
I am trying to construct a CPU using only the IC library but I can't seem to figure out how to implement a reset circuit. I don't want to necessarily modify the IC subcircuits because I am trying to be as pure as possible, but every IC complains if the VCC power is not constantly active. Is there another way to control the starting values without having to tamper with the IC subcircuits?

hneemann, 2022-04-05:
I'm not sure I understand what you are trying to achieve. But there is a reset component with only one output, which changes its state when the circuit has stabilized after switching on (after starting the simulation).

Ka3oose, 2022-04-08:
Well, I managed to stabilize the bus using pull down resistors, but what I was wanting to achieve is the creation of a diagram using the 74XX chips that I could use to prototype a real circuit someday. Since I will need debounce/reset circuits in real life, I wanted to simulate them as much as possible in the simulator as well. I do know about your reset component, and I have used it before. But what I want to do is essentially create a circuit that would represent that reset component and then use it with the ICs.

Ka3oose, 2022-04-08:
Understood, I'll stick with what's available and make do! Thank you!

closed 2022-04-08 by Ka3oose

prlaba, 2022-04-17:
A somewhat related question:  Digital’s Monoflop component would make for a great “power-on” reset signal, but what’s its trigger?  Have you considered some sort of “simulation started” signal that could be used to trigger the monoflop (or any other component)?

For those of use using Digital to simulate a CPU or entire system, having such a signal triggering a monoflop would approximate common power-on reset circuits like the Dallas DS1813 chip.

hneemann, 2022-04-17:
There is a reset component ("misc/Reset") that essentially acts like a DS1813. 
There is no need for a monoflop. 
The reset component generates the reset signal when the circuit has stabilized after switching on (simulation start). 
So exactly when the state of the circuit would not change any more, no matter how long you would wait.

prlaba, 2022-04-18:
Ah, I missed that one.  Thanks!

I did come up with a circuit that generates a PWR_ON_RESET signal.  The signal remains high for a fixed number of CLK cycles, then goes low forever (until the simulation is reset).  But your Reset component eliminates any need for that.