hneemann/Digital #950: No output connected to a wire ([3; Splitter, In_1; XOr]). The state of the wire is undefined.

656e7269717565, 2022-04-14:
I'm doing a circuit that would be capable to encrypt in AES 128, I'm doing a part of the "mixcols" process and this error keeps appearing, how can I fix it
[image]

hneemann, 2022-04-14:
could this be a duplicate of #337?

656e7269717565, 2022-04-14:
so is there a way I can flip the thing or do I have to rethink to make something that would give the same output

hneemann, 2022-04-15:
The splitter at the bottom left appears to have eight inputs and an 8-bit output. 
You can simply configure this splitter to have eight outputs and one 8-bit input.

656e7269717565, 2022-04-15:
I've designed the other chips with byte i/o, is there a way to split wires because it's going to be a mess to redo everything to make it bit i/o

hneemann, 2022-04-15:
I'm not sure I understood the question, but that is exactly the purpose of the splitter.

656e7269717565, 2022-04-15:
nevermind, I just realized I could look at one of the chips of an EEPROM to see how it splitted the connections, you shouldn't give the "splitter/merger" that name as it only merges inputs and the name suggests that it could Split inputs (:

hneemann, 2022-04-15:
The splitter is able to split and merge wires:

[image]

[z.dig.zip](https://github.com/hneemann/Digital/files/8497420/z.dig.zip)

656e7269717565, 2022-04-16:
why did I think output splitting meant the number of bits **one** output would have XD, I'm just dumb, anyway thanks for solving my problem xd

closed 2022-04-16 by hneemann