hneemann/Digital #952: Windows 10 Unusable because of graphics glitches

Surac, 2022-04-22:
version: v0.29

Using the program i discovered graphic gliches on windows 10

[image]

using Java jre-8u331-windows-i586.exe, no Display scaling, using Digital_noD3D.exe

is there a way to make it usable?

hneemann, 2022-04-22:
It appears to be a duplicate of #227.

Please try a more recent version of Java.

Surac, 2022-04-22:
Error is gone with JAVA "jdk-17_windows-x64_bin.exe" from Oracle

p.s. Starting Digital with no SE or JDK installed redirects you to the download page of oracle for Java 8. That was my fault

hneemann, 2022-04-22:
> Starting Digital with no SE or JDK installed redirects you to the download page of oracle for Java 8.

I'm not sure if that was really your fault.

closed 2022-04-22 by hneemann