hneemann/Digital #957: TCP/IP does not appear to work

FilipSzkandera, 2022-05-17:
I have downloaded the latest Digital version as well as the Assembly program and they are not communicating at all.
I also tried a Python program with socket library and it sees the port, but again I'm not able to send any commands - or I can send them, but Digital won't do anything (from what can I observe) and I do not get any response back.

Could you please verify that this feature still works?
Thanks.

My python code:

```
import socket

HOST = "127.0.0.1"  # The server's hostname or IP address
PORT = 41114  # The port used by the server

with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
    s.connect((HOST, PORT))
    # command = "start:C:/Users/xxx/Downloads/Digital/app.bin".encode("utf-8")
    command = "step".encode("utf-8")
    s.sendall(command)

    data = s.recv(1024).decode("utf-8")

print(f"Received {data!r}")
```

hneemann, 2022-05-17:
Everything works as expected on my computer.
However, in the latest version, the TCP/IP server must be enabled in the settings. On some machines there were problems when a port is opened. 
But looking at your code is notected that the length of the string is missing. See [here](https://github.com/hneemann/Assembler/blob/ffed70125764ac286bd3cbd048a807f33e9885b5/src/main/java/de/neemann/assembler/gui/RemoteInterface.java#L101).

FilipSzkandera, 2022-05-17:
Oh, okay then. Still I'm struggling with the interpretation of what should I send and with no response from Digital. Could you please look at this code?

```
import socket

HOST = "127.0.0.1"  # The server's hostname or IP address
PORT = 41114  # The port used by the server

def strlen_as_bytes(string):
    return bytes(len(string) >> 8), bytes(len(string) & 0xff)

with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
    s.connect((HOST, PORT))

    command = "run"

    s.sendall(strlen_as_bytes(command)[0] + strlen_as_bytes(command)[1] + command.encode("utf-8"))

    data = s.recv(1024).decode("utf-8")

print(f"Received {data!r}")

```

Thank you.

hneemann, 2022-05-18:
To by honest, i have no idea what happens if byte+byte+array is evaluated in Python.
But i would expect that [(byte+byte),array...] is generated instead of [byte, byte, array...]?

FilipSzkandera, 2022-05-18:
Sure, I'm really just guessing what it wants, what [this](https://github.com/hneemann/Assembler/blob/ffed70125764ac286bd3cbd048a807f33e9885b5/src/main/java/de/neemann/assembler/gui/RemoteInterface.java#L10) really means.
Lets say I want to send the command:
`run`
So the first two bytes will be the length of the string (in this case 0x00, 0x03) and after that there will be the string in the byte format, so 0x72, 0x75, 0x6E - ("run" in hex). Is this correct?

Thanks.

hneemann, 2022-05-18:
That correct! If you want to send the `run` command, you have to send 

`0x00,0x03,0x72,0x75,0x6E`

But keep in mind: With UTF8 there may be characters that occupy two or more bytes. You must specify the number of bytes that are sent, not the possibly smaller number of characters.

FilipSzkandera, 2022-05-18:
Awesome, it now works!
If this will be useful to anyone else, here is a working Python demo (just for the start function, but that can be changed to anything else):

```
import socket

HOST = "127.0.0.1"  # The server's hostname or IP address
PORT = 41114  # The port used by the server

def len_and_str_to_bytes(string: str):
    upper_byte = bytes([len(string) >> 8])
    lower_byte = bytes([len(string) & 0xff])

    result = upper_byte + lower_byte + string.encode("utf-8")

    print("Sending: ", result)
    return result

with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
    s.connect((HOST, PORT))
    s.settimeout(5)

    command = "start:C:\\Users\\xxx\\Downloads\\Digital\\app.bin"
    

    s.sendall(len_and_str_to_bytes(command))

    data = s.recv(2)
    data = s.recv(1024).decode("utf-8")

    s.close()

print(f"Received {data!r}")
```

One more question remains, but I will start another thread about that.
Thank you a lot!

closed 2022-05-18 by FilipSzkandera