hneemann/Digital #960: Inconsistent step function in TCP/IP protocol

FilipSzkandera, 2022-05-25:
If I send a `step` packet via the TCP/IP protocol (and we are assuming that Digital simulator is running), the clock signal will go high and low instantly. But when I manually set the clock to high and then call the `step` function, it will just go low. 

Much more useful would be if the `step` will just set the clock to high or low, without the full cycle. But in order not to disrupt any programs that  are based on this, I would suggest adding a `half_step` or something similar.

Thank you very much.

hneemann, 2022-05-25:
Hmm! 
If you want to examine the function of the processor in such detail, maybe the assembler IDE is not the right tool for the job?

FilipSzkandera, 2022-05-25:
I have my own simulator of the same CPU written in C++ and the Digital version is not working exactly right, so I just connected these two together via TCP, so I'm not really using the IDE at all.

hneemann, 2022-05-25:
That's an interesting use case! I have to think about it!

FilipSzkandera, 2022-05-25:
That would be really awesome and time saving! Not sure if I should continue or wait for that feature, because that would help a lot more. 

And by the way a big thanks for these features, there are really amazing!

closed 2022-05-27 by hneemann

hneemann, 2022-05-27:
@FilipSzkandera 
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

FilipSzkandera, 2022-05-27:
It works perfectly, thank you very much.

commit 81643abe (2022-05-27): adds a clock command to the remote interface; closes #960