hneemann/Digital #966: 7493 Counter error

pazuzu72, 2022-06-14:
I am trying to use a 7493 (counter) for one of my projects but I always get the error: "Cannot analyse Node FlipflopJKAsync".
What could be the problem? With other 74XX everything goes smoothly.

hneemann, 2022-06-14:
This is not a bug 
The circuit analyzer used in this simulator is not capable of analyzing an aysnchronous JK flip-flop.

pazuzu72, 2022-06-14:
Thaks for your answer.
So I can't use it inside an ATF150x? Or is there another way?

hneemann, 2022-06-15:
The ATF1504 itself supports asynchronous set and clear. It is only the simulator that does not support the corresponding analysis. The simulator uses a textbook-type analysis and synthesis technique. Not something comparable to what HDL based tools like ISE or similar use.
The focus is not to support all features of a chip, but to make what is typically used in teaching run on a real chip.

pazuzu72, 2022-06-15:
Ok, thank you very much!

closed 2022-06-27 by hneemann