hneemann/Digital #968: 74194 Pinout Mis-numbered

tcwan, 2022-06-16:
I encountered this issue recently when trying to use it to implement the Ring Counter example (from "Fig. 8-42, J. F. Wakerly, Digital Design: Principles and Practices, 4th Edition, Pearson, 2014")

Actually the issue is not that the logic was implemented incorrectly in the previous verison (release v.029, 2-12-2022). The issue was that the numbering of the pins Q0-Q3 was in the opposite order than that defined in the datasheet. In the datasheet, QA (pin 15) == Q3 (pin 12), while QD (pin 12) == Q0 (pin 15). Similarly for inputs A-D. If you rearrange the pin ordering for the 74194 part in the simulator to follow the pin ordering for the datasheet, it will work correctly. 

TLDR: The pin labels in the simulator are correct, it is the pinouts that are wrongly placed for D0-D3 and Q0-Q3.

_Originally posted by @tcwan in https://github.com/hneemann/Digital/issues/921#issuecomment-1157083210_

tcwan, 2022-06-16:
This is a continuation of the discussion for [74194 pre-release fix](https://github.com/hneemann/Digital/issues/921). Figured I should open a new issue in case the previously closed issue was buried.

tcwan, 2022-06-21:
> Have you tried the latest [pre-release](https://infdigital.dhbw-mosbach.de/)?
The commit #0e080710e989c69c83406085a6a395a875e3f622 by coreiddc mentioned in the linked issue?

Yes, I have tried it. It swiched the S0 and S1 behavior and does not conform to the datasheet behavior. As I've mentioned, the issue is the pin numbering not the behavioral logic of the original part.

hneemann, 2022-07-31:
I am pretty sure that the pin numbering is correct. 
See 74194real.dig for comparison.

closed 2022-07-31 by hneemann

tcwan, 2022-08-09:
I'm not sure where 74194real.dig is located.

I've enclosed a 4-bit left-shifting ring counter circuit using 74194 as an example.

Normally Q3,Q2,Q1,Q0 assumes that Q3 is the MSB and Q0 is the LSB.
For the case of the 74194, the outputs are labelled QA,QB,QC,QD where QA is the MSB and QD is the LSB.

The left-shifting ring counter circuit needs to Toggle RESET to load a '1' into the circuit, after which it will circulate normally (SER->QD->QC->QB->QA->QD).

The sequence should go Pin 12 -> Pin 13 -> Pin 14 -> Pin 15 -> Pin 12.

In Digital, the 74194 model loads SER into Pin 15 -> Pin 14 -> Pin 13 -> Pin 12, which is the opposite direction.

Basically the logic behavior is correct, but the Pin assignment for Q3 to Q0, and D3 to D0 is not.

[74194-4bit-RingCounter.dig.zip](https://github.com/hneemann/Digital/files/9285837/74194-4bit-RingCounter.dig.zip)
[image]
[image]
[image]
.

hneemann, 2022-08-09:
To be honest, I don't get it. 
The two 74194 circuits behave exactly the same. 
To ensure this, I have extended the test case parser so that not only the names of the pins can be used, but also the pin numbers. The test case based on pin numbers works unchanged in both circuits.
But in fact the pin names are not the same in all datasheets of the different manufacturers. Some pin names are chosen more wisely by one manufacturer than by others. But that's the way it is. 
In the end it is only the pin numbers that matter for the 74xx circuits, not the pin names used in the various data sheets.

tcwan, 2022-08-26:
> But in fact the pin names are not the same in all datasheets of the different manufacturers. Some pin names are chosen more wisely by one manufacturer than by others. But that's the way it is.
In the end it is only the pin numbers that matter for the 74xx circuits, not the pin names used in the various data sheets.

I have not seen any datasheet which uses Q0-Q3 naming, so I defer to you in that regard. 

Logically both circuits work as expected, but the wiring to the specific pin numbers are not the same (i.e., the two DIL packages defined in Digital are not drop-in compatible on a PCB if it were a real device).

commit 35b23175 (2022-07-31): added a 74194 data book implementation, closes #968