hneemann/Digital #983: Crashing with OpenJDK 18.0.1.1 (2022-04-22)

ShifatHossain, 2022-07-13:
Tried to open Digital using OpenJDK 18.0.1.1 (OpenJDK runtime environment 18.0.1.1+2-6) using the following command in Windows 11:

`java -jar Digital.jar`

And got the following error traceback:

`Exception in thread "main" java.lang.ExceptionInInitializerError
        at com.thoughtworks.xstream.XStream.setupConverters(XStream.java:832)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:574)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:496)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:465)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:411)
        at com.thoughtworks.xstream.XStream.<init>(XStream.java:378)
        at de.neemann.gui.language.Bundle.getxStream(Bundle.java:23)
        at de.neemann.gui.language.Bundle.<init>(Bundle.java:37)
        at de.neemann.digital.lang.Lang.<init>(Lang.java:89)
        at de.neemann.digital.lang.Lang.<init>(Lang.java:20)
        at de.neemann.digital.lang.Lang$InstanceHolder.<clinit>(Lang.java:25)
        at de.neemann.digital.lang.Lang.get(Lang.java:36)
        at de.neemann.digital.gui.Main.<clinit>(Main.java:104)
Caused by: java.lang.reflect.InaccessibleObjectException: Unable to make field private final java.util.Comparator java.util.TreeMap.comparator accessible: module java.base does not "opens java.util" to unnamed module @46238e3f
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:354)
        at java.base/java.lang.reflect.AccessibleObject.checkCanSetAccessible(AccessibleObject.java:297)
        at java.base/java.lang.reflect.Field.checkCanSetAccessible(Field.java:180)
        at java.base/java.lang.reflect.Field.setAccessible(Field.java:174)
        at com.thoughtworks.xstream.core.util.Fields.locate(Fields.java:40)
        at com.thoughtworks.xstream.converters.collections.TreeMapConverter.<clinit>(TreeMapConverter.java:50)
        ... 13 more`

hneemann, 2022-07-13:
Which Digital version are you using?

ShifatHossain, 2022-07-13:
I just tried the latest release (v0.29, Build Time: 2022-02-11 17:13) and it ran on OpenJDK 18.0.1.1 without any issues. 

Sorry to bother you. I will close this thread, since this issue is resolved already.

closed 2022-07-13 by ShifatHossain