hneemann/Digital #987: How to load program from file

CyberGai, 2022-07-16:
I read through the documentation and found out about loading a program file into memory for processors. I went ahead marked the ROM I am using in my circuit as Program Memory. I then went to `Edit -> Circuit Specific Settings -> Advanced -> Preload program memory at startup` and also located the file in the `Program file` setting under it. The circuit and and program file are in the same directory. When I start my simulation, the data that is loaded into memory is not the one I had written in the program file.

Following is the content of my program file named as hex.txt:
```
0x00600293
0x00007317
0x00600393
0xFE728CE3
```

Following is the output program I am receiving in the simulation
```
0x30307830
0x32303036
300A3339
0x30303078
0x31333730
78300A37
0x30363030
0x3339330
4678300A
0x38323745
A334543
A
0
0
0...
```

I think I'm doing something obviously wrong here. Should the program file extension be txt? Or the maybe the instruction syntax is wrong?

Aaron-Rumpler, 2022-07-17:
Digital reads the file specified as a binary file, whereas you've got the data you want written in hex in a plain text file. The output you're seeing is the contents of `hex.txt` file itself (the character `0` is `0x30` in hex). I'd suggest using a hex editor to create your file.

CyberGai, 2022-07-17:
Oh, so the file to be selected should be a binary file. Thank you for the guidance.

closed 2022-07-17 by CyberGai