hneemann/Digital #988: Missing Digital features

prlaba, 2022-07-16:
I'm a long-time Logisim and Logisim-Evolution (L-E) user.  I've been following and playing with Digital for a while now, and am very impressed with its many features and cleaner architecture compared to Logisim and L-E.

But there are three "missing" Digital features that prevents me from migrating to Digital:

1. Cannot pause/resume a simulation.  Once I start a simulation, my only option is to stop it.  But stopping the simulation immediately resets the simulation, losing all state.

L-E allows me to freely start and stop the simulator without losing state (in fact I don't even need to stop it, as I can examine the state of any circuit or sub-circuit at any time).

How difficult would it be to add Pause and Resume buttons?  Pausing the simulation would stop the simulation but still maintain state; resuming the simulation would restart the simulation at the same state.

I understand that Digital only allows a circuit or component to be modified while the simulation is stopped, so this new _paused_ state would raise the obvious question as to whether a circuit could be changed while the simulation was paused.  It would be acceptable to disallow changes while the simulation was paused.

2) Viewing the state of sub-circuits while the simulation is running.  My L-E circuits are complex, often containing a dozen or more sub-circuits.  When testing, it's critical that I be able to "drill down" to see the state of the various parts of a sub-circuit.  I can't imagine trying to debug a complex circuit without this ability.

Would this be a significant change to Digital's architecture to be able to view the state of a sub-circuit while the simulator is running?

Coupled with the Pause/Resume feature, it would be ok to have to pause the simulation to view a sub-circuit's state, so long as that state wasn't lost.

3) The Clock Input component's _Start real-time clock_ attribute can only be changed when the simulation is stopped.  Coupled with the lack of Resume feature, that means I can't start and stop the clock during simulation.

I often need to run through many clock cycles before stopping the clock to view a circuit's state.  L-E provides icons for stepping the clock one full clock cycle, one-half cycle or running the clock in real time, so I can easily start or stop the clock as needed.

With Digital I can toggle the clock's state while the simulation is running, but I can't start of stop the clock.  Digital does not allow any component's attribute to be changed while the simulator is running, so I can't toggle the Clock Input component's _Start real-time clock_ attribute unless the simulator is stopped.  And without the Pause/Resume feature mentioned above, there's no way to do that without restarting the simulation and losing state.

How difficult would it be to treat the _Start real-time clock_ attribute as a _simulation control_, rather than a Clock Input component attribute?  It would obviously apply to all Clock Input components.

I'd appreciate hearing your thoughts on these three "issues".

Thanks and keep up the great work!  Digital keeps getting better because of you (I wish there was another "you" maintaining L-E).

hneemann, 2022-07-18:
1. Stop and continue seems like a simple addition. But what is the use case? When the clock is running, it is difficult to pause the clock at a particular circuit state. The "Break" component is a much simpler solution for investigating a specific state.
2. The implementation of such a feature is a complicated task, because the circuit is only a kind of factory data structure to create an executable model from it. There is a lot of bookkeeping to implement to infer from the model back to the original circuit. From my point of view, the benefit does not justify the expected effort and the increased complexity.
3. Due to the described model generation it is difficult to let a subsequent change of the circuit influence/change the already generated model.

In my opinion, testing by "drilling down" a circuit is not a good practice. It is better to add test cases to the subcircuits. This is exactly the same as in software development: testing by adding and deleting or commenting out many print statements is not a good idea, writing unit tests is the better way.

hneemann, 2022-07-28:
see #84

hneemann, 2022-07-28:
see #1005

closed 2022-07-28 by hneemann