hneemann/Digital #991: Digital.exe requires Java 8

Likqez, 2022-07-22:
After downloading `v0.29` and executing `Digital.exe` I get prompted with a message that tells me that JRE 8.0 is required. Although your README states:

> A [Java Runtime Environment](https://www.java.com/) (at least JRE 8) is required to run Digital.

And a JRE higher than 8.0 is installed.
Running the jar file directly via ``java -jar Digital.jar`` successfully launches the application.

#### Installed Java Version:
```
PS C:\Users\lanzn> java -version
openjdk version "17.0.2" 2022-01-18
OpenJDK Runtime Environment (build 17.0.2+8-86)
OpenJDK 64-Bit Server VM (build 17.0.2+8-86, mixed mode, sharing)
```
Is there a possibility to correct the executable?
or/and can the requirement get raised to current LTS.

hneemann, 2022-07-22:
Would you please try the latest [pre release](https://infdigital.dhbw-mosbach.de/)?
I don't have a Windows machine at hand.

I have modified the download link to https://adoptium.net.
A long overdue modification.

Likqez, 2022-07-22:
I tested ``v0.29-43-g729d38d``.
Still same error popup requiring JRE 8.
Noticed the changed Download link.
Additional windows defender popup, but I guess that's because of the pre-executable, which isn't signed or something like this.

hneemann, 2022-07-22:
Hmm! Would you like to unzip and test the attached exe file?

[Digital.exe.zip](https://github.com/hneemann/Digital/files/9166210/Digital.exe.zip)

vdelau, 2022-07-22:
I ran into the same issue, the version mentioned in the last comment worked for me. I noticed this file is a lot bigger?

hneemann, 2022-07-22:
@vdelau 
Could you pleace start the exe file from the command line and attach the console output to this issue?

vdelau, 2022-07-22:
When running any Digital.exe executable from PowerShell or cmd I do not get any output in the console (tried release, pre-release and the above executable). Pre-release build also complains about Java version, with the Adoptium URL.

```
C:\path\to\Digital>Digital.exe

C:\path\to\Digital>java --version
openjdk 18.0.1.1 2022-04-22
OpenJDK Runtime Environment (build 18.0.1.1+2-6)
OpenJDK 64-Bit Server VM (build 18.0.1.1+2-6, mixed mode, sharing)

C:\path\to\Digital>ver

Microsoft Windows [Version 10.0.22000.739]

C:\DEVELOPMENT\Digital>set|find "Java"
CLASSPATH=.;C:\path\to\Java\jdk-18.0.1.1\lib;C:\path\to\Java\jdk-18.0.1.1\jre\lib
JAVA_HOME=C:\path\to\Java\jdk-18.0.1.1
JDK_HOME=C:\path\to\Java\jdk-18.0.1.1
JRE_HOME=C:\path\to\Java\jdk-18.0.1.1\jre
Path=<snip>;C:\path\toJava\jdk-18.0.1.1\bin;<snip>
```

hneemann, 2022-07-22:
@vdelau 
Thanks for the effort, but this is hard to believe. 
The exe file attached to this issue contains lots of log outputs even when everything is working. 
Hmm! No idea what's going on.

hneemann, 2022-07-22:
@vdelau 
Whould you please try this updated exe file?

[Digital.exe.zip](https://github.com/hneemann/Digital/files/9169617/Digital.exe.zip)

Likqez, 2022-07-23:
I can confirm, the latest executable does work. Thanks 👍

Johnlon, 2022-09-15:
Even with JDK 8 on the path and JAVA_HOME set I also get the java8 required message.
The zip above does work though.
Why bother with an exe at all though, ,would a bat do whatever's needed?

hneemann, 2022-09-16:
@Johnlon 
The simulator has its own font scaling option in the settings.
Maybe you have told the operating system to increase the font size and also increased the font size in the app itself?

Likqez, 2022-09-16:
Is there a release date scheduled, which includes the fix for this issue?

Johnlon, 2022-09-16:
Literally just installed it on a 4k dell laptop and the fonts showed as in the attachment. I used the os settings to correct this as I've had to do that once or twice before. I understand there is also a setting in the app, but I don't know anything about that.

hneemann, 2022-09-16:
@Johnlon 
Make sure that the font is not enlarged twice: Once by the OS and once by the app.
In the settings you can adjust the font size: Edit->Settings->Advanced->Font size.
What value is set here?

Johnlon, 2022-09-16:
@hneemann menu size says 100 - this is straight after unzipping

hneemann, 2022-09-16:
@Johnlon 
Hmm! Is it possible to disable the font enlargement of the operating system?

Johnlon, 2022-09-16:
@hneemann on the other ticket ..https://github.com/hneemann/Digital/issues/1027#issuecomment-1248945551
I included what I did on win10 to make the fonts smaller. This was on a clean install of the product.
If I undo this setting then I expect massive fonts again on my 4k screen. I've not adjusted any app settings.

kosch104, 2022-10-30:
@hneemann Referencing the original topic of this issue:
Could you create an updated release with the functioning new exe? 
Opening files directly by extension does not work with the new exe, would be great if you had time to implement that!

If not I'd be happy to try implementing it myself and creating a PR, if you'd be interested in that

Likqez, 2023-02-14:
@hneemann is the fix already included in the latest release?

ProxyPlayerHD, 2023-08-21:
even on the newest release it says "this application requires JRE 1.8.0" with both PATH and JAVA_HOME being correct (ie linking to my JRE 1.8.0 install), and it also links to the oracle site, i thought that was supposed to be changed?
on a side note it would be nice if there was an extra button next to "OK" on that window that when clicked opens the website, instead of it always opening when you close the window... as that got really really annoying when trying to experiment and find out why digital wouldn't launch and having to close a browser tab each and every time.

the exe downloaded from this issue works but the extra debug output opens an extra window each time that needs to be closed manually. so yea a fix on the newest release would be nice.

>Johnlon: Why bother with an exe at all though, ,would a bat do whatever's needed?

well i don't know if you can tell windows to open a specific file type with a bat file. but it does work with exe's.
so you can simply double click `.dig` files and have them open directly in digital.

though ironically the exe file from this issue cannot do this... when i launch it directly (by double clicking Digital.exe or through cmd) it works fine, but when launched by opening a .dig file it again complains that it needs JRE 1.8.0.

godammit. this all worked perfectly fine like a week ago, all i did was remove oracle's java and replace it with Eclipse Adoptium openJDK and now everything is broken :(
i might just install oracle java again and see if that fixes things