hneemann/Digital #995: Ordering for ABC-inputs for the 74238 is wrong in simple mode

mengstr, 2022-07-25:
Seems to be the same problem as in #925

hneemann, 2022-07-26:
To be honest, when I created most of the 74xx chips, either there was no simple-mode yet, or I didn't consider that it existed and someone could use it.
Now in retrospect, changing the pin order is difficult because any change breaks all existing circuits that use this chip.

mengstr, 2022-07-26:
Yes, I understand the backwards-compatibility problem.  

What happens if I unlock a library IC and modifies it? Will the changes be saved in the common library or just specifically in my design?   Without trying (which would be easy enough) I guess that it will be saved in the library. So the safest thing to do would be to just copy the designs and make my own versions of them when needed.

hneemann, 2022-07-26:
The library is just a folder on your PC. This folder is searched at startup and the library menu is created accordingly.
It is possible to change the circuits in this folder or add new ones. Locking the circuits does not affect its function. It serves only as a protection against unintentional modification.
But it is also possible to copy the library folder and select the copy as the new library folder in the settings.

mengstr, 2022-07-26:
I was thinking of the case where the design was copied to a another computer. Then all the changes to the library part would be reverted back to the original version rendering the design non-working.  

So the modified ICs should be probably be copied as Embedded Circuits to ensure the modifications are a part of the actual design.

hneemann, 2022-07-26:
In this case it is better to rename the circuits and put them in a subfolder of your design. If there are several circuits with the same name, the name will not be unique and an error will occur.
In digital, the path of a circuit does not matter, identification is based solely on the filename.

mengstr, 2022-07-26:
Ah, I see... I'll do it that way then.  Thanks.

closed 2022-07-26 by mengstr