hneemann/Digital #1005: adding a pause button

hneemann, 2022-07-28:
Add a pause button to pause/resume the running clocks.

Johnlon, 2022-09-15:
Yes please - really needed

prlaba, 2023-01-20:
I assume this is not trivial to implement.

A somewhat related feature would be to be able to see the state of a subcircuit -- any number of levels deep -- when the simulation is paused.  By state I mean show the state of each input, output and wire in the subcircuit.  It's an almost-required feature when troubleshooting complex circuits with multiple levels of subcircuits.

Logisim and Logisim-Evolution provides this capability natively.  In fact it's not even necessary to pause the simulation to see (or change) the state of the main circuit or any of its subcircuits.  You can open a subcircuit and watch it change in real time, as if it were the main circuit.

There are so many great features in Digital -- features that put it head and shoulders above other simulators like Logisim and L-E.  But not being able to  view and troubleshoot subcircuits while the simulator is running (or paused, if that feature existed) is the main reason I haven't been able to migrate my more complex projects over to Digital.

Johnlon, 2023-01-20:
I imagine it's just an event loop that needs a pause. It's not sensitive to walk clock after all