hneemann/Digital #1011: MIDI component does not work on Artix Linux.

RadiantUwU, 2022-08-02:


hneemann, 2022-08-02:
I assume that the problem lies in the configuration of your Linux.

RadiantUwU, 2022-08-02:
Update: The Plasma DE doesnt even show that Digital is even trying to play sound

RadiantUwU, 2022-08-02:
> I assume that the problem lies in the configuration of your Linux.

I would doubt that since other Java programs work fine (for example Minecraft)

hneemann, 2022-08-03:
> I would doubt that since other Java programs work fine (for example Minecraft)

That may well be, but I doubt that these programs use a MIDI syntheziser. :smile:
MIDI synthesizers are not used much these days because playing WAV files directly is no longer a problem and produces a more consistent sound because you don't have to rely on MIDI sound tables.

RadiantUwU, 2022-08-03:
Alright, i attempted to run this example code for testing purposes, it played no sound without debugging mode present on InteliJ
```java
import javax.sound.midi.*;

public class MidiSynthesizerSample {
    public static void main(String[] args) {
        int[] notes = new int[]{60, 62, 64, 65, 67, 69, 71, 72, 72, 71, 69, 67, 65, 64, 62, 60};
        try {
            Synthesizer synthesizer = MidiSystem.getSynthesizer();
            synthesizer.open();
            MidiChannel channel = synthesizer.getChannels()[0];

            for (int note : notes) {
                channel.noteOn(note, 50);
                try {
                    Thread.sleep(200);
                } catch (InterruptedException e) {
                    break;
                } finally {
                    channel.noteOff(note);
                }
            }
        } catch (MidiUnavailableException e) {
            e.printStackTrace();
        }
    }
}```

RadiantUwU, 2022-08-03:
Huh, not it doesnt play sounds anymore even with debugging mode

RadiantUwU, 2022-08-03:
Now it does, weird stuff

closed 2022-08-09 by hneemann