hneemann/Digital #1014: Add Program Clock

AsparaGus116, 2022-08-04:
This is not a bug report, but it would be nice if one could implement a program clock. What I mean is, if you have one clock in a simulation, you can press 'c' to step through one cycle of the clock. In my simulation, I have multiple clocks, and it doesn't work with multiple. Would it be possible to implement one clock being the "Program Clock", where that would be the one to be stepped through? It might not be possible, because it would have to interact with other clocks, but it would be very nice if it worked.

hneemann, 2022-08-04:
@mengstr 
You can use the break component to detect the completed instruction. 
With "RunToBreak" you could then execute complete instructions.

mengstr, 2022-08-04:
Hm... I'll try that the next time. That will probably make this quickly cobbled together "step & clock phase" block a bit simpler. 
[image]