hneemann/Digital #1022: the order of inputs and outputs get messed up when I save as a chip/embedded circuit

metoshinakamoto, 2022-09-06:
attached is the picture of the chip along with the inside of it, notice how the order is completely different

[image]
[image]

hneemann, 2022-09-06:
Use "Edit->Order Inputs" and "Edit->Order Outputs" to order the inputs and outputs accordingly.

xotmatrix, 2022-09-06:
That's probably the best option but you could also use:
* From the parent circuit: "Right-click Adder -> Shape: Layout"
* From the embedded circuit: "Edit -> Circuit specific settings -> Advanced -> Shape: Layout"

metoshinakamoto, 2022-09-07:
👍

closed 2022-09-07 by metoshinakamoto