hneemann/Digital #1029: Customing gates without programming

iijunxz, 2022-09-18:
I was trying to implement the fredkin gate in Digital, and I had tried these two methods:

1. Embedded circuits, but it seems unable to change data bits dynamically.
2. Lookup tables, but it can only accept 1-bit of input.

Apart from use java to add new componets, is there are any easier way to add simple gates?

hneemann, 2022-09-18:
But what's wrong with an implementation like this?

[image]

iijunxz, 2022-09-18:
> But what's wrong with an implementation like this?
> 
> [image]

That may be my fault, but I haven't found out how to change the data bits of embedded circuits. For this example, I hope every gate can be replaced by Fredkin gate, therefore I need multiple gates in 1bit, 8bits, 32bits etc.
If there isn't a solution yet, then I have two ideas:
1. Add a bit-by-bit LUT, which simulate a truth table on each bit.
2. Add a way to transfer variables into embedded circuits, and enable to use them as a number (Not sure whether technically feasible or not)

hneemann, 2022-09-18:
I think I didn't understand your question at first.
Your second suggestion already exists. It's called a generic circuit.
Here is a Fredkin gate as an example on how to use generic circuits: [Fredkin.zip](https://github.com/hneemann/Digital/files/9594008/Fredkin.zip)

iijunxz, 2022-09-18:
That's what I was asking for, and thank for replying.

closed 2022-09-18 by iijunxz