hneemann/Digital #1044: Adding delays to exported verilog netlist

eliask05, 2022-10-22:
Is it possible to add a delay to a component (such as an and gate) that gets exported to the verilog netlist?

In other words, instead of generating "assign x = (a & b)" generate "assign #1 x = (a & b)". I understand that if the intended purpose of the netlist is FPGA implementation, this is meaningless. It would be however nice to be able to add such delays when the intention is just to simulate for more realistic results.

In fact, if full delay specification, such as specify blocks, was implemented, that would be really nice!

I know I'm asking too much but...

Thanks!

hneemann, 2022-12-14:
To be honest, the complete HDL export was only implemented to give students an easy way to run a circuit designed in the simulator on an FPGA. It was not even intended that a student would open the generated files at all, or even run the HDL code in another simulator to study its behavior.
I don't think the simulator can ever become a viable tool for generating HDL code. The possibilities of an HDL are much too extensive for that and the possibilities of the simulator are too limited in comparison.

eliask05, 2022-12-14:
I fully understand. On the other hand, to the best of my knowledge -- please correct me if I am wrong, Digital is the only free tool that can convert a schematic to Verilog. This can be very helpful, especially with messy schematics with no clear repeating structure.

As always, thank you for the wonderful tool!