hneemann/Digital #1048: request: Add registers with increment option

Hamid1382, 2022-10-29:
Hi thanks for powerful app but a small problem i have encountered is that there is no register with INC input that increments data stored by one , in computer architecture they are very useful. until then i have to implement inc by myself. tnx again

hneemann, 2022-10-29:
But it is not a big deal to build such a component as a sub circuit.
It is hardly worth to add something like this as a built-in component.

TheWildJarvi, 2022-10-29:
Use the built in up down counter and set it up to increment or load a value.

hneemann, 2022-10-30:
@TheWildJarvi 
That's right! I didn't think of that at all! :smile: 
The built in up down counter is essentially a register with inc and dec input.

closed 2022-10-30 by hneemann

Hamid1382, 2022-11-05:
> But it is not a big deal to build such a component as a sub circuit. It is hardly worth to add something like this as a built-in component.

I have to use up to 128 registers and i haven't seen any button like create Integrated circuit

hneemann, 2022-11-05:
Copy the attached file to your circuit folder and import the register with _Components->Custom->incReg_.
Then set the Bits value accordingly.

[incReg.dig.zip](https://github.com/hneemann/Digital/files/9943867/incReg.dig.zip)