hneemann/Digital #1086: Request: Bargraph

krangerich, 2022-12-29:
It would be nice to have a bargraph component. 

While you could use single LEDs instead, a bargraph has its advantages: 
- You only have add one component instead of like 8
- Takes less space, because you have to place single LEDs every 2 grid lines. Inputs of a bargraph should be 1 grid line apart.
- It just looks nicer, e.g. when visualizing CPU registers

Number of bars should be configurable (e.g. 4, 8, 10, 16)

xotmatrix, 2022-12-29:
One can change the size of the LED so that it only takes up one grid space. Constructing a group of LED is not that difficult but a bar graph display could be convenient. Alternatively, I sometimes wish it was possible to define groups of components to make selecting all of them easier.

[image]

hneemann, 2023-01-09:
Since the suggestion of @xotmatrix is sufficient in my opinion, I close this issue.

closed 2023-01-09 by hneemann