hneemann/Digital #1090: Improving the reproducibility of the build process

Dettorer, 2023-01-08:
Hi! I'm trying to package Digital for NixOS, for this I need a way to make the build process completely deterministic/reproducible. Following the [maven documentation on reproducibility](https://maven.apache.org/guides/mini/guide-reproducible-builds.html), I managed to make the build deterministic by setting the recommended `project.build.outputTimestamp`[^1] property on the cmdline and patching `pom.xml` to use a fixed value for the `Build-Time`[^1] manifest entry of the assembly plugin[^2] and to upgrade the `jar` and `assembly` plugins to their first version that support build reproducibility (3.2.0 in both cases as recommended by `mvn artifact:check-buildplan`).

[^1]: I used the timestamp of the commit I build from (v0.29 for now) every time I need a fixed date.
[^2]: I'm not sure why only this one is needed, the jar plugin also has a `Build-Time` manifest entry I didn't have to patch it.

Patching on NixOS' packaging side is a viable solution, but to make the packaging job easier, would you be inclined to 1) upgrade the plugins here too, and 2) add a pom.xml property that can override all `Build-Time` entries from maven's command line directly? I can make a PR that implements those two points. Do you see any problem with my approach? I'm not experimented with maven at all.

Thanks btw for implementing https://github.com/hneemann/Digital/issues/289#issuecomment-513721481, packaging for NixOS was way easier without having to make the full git history available at build time.

hneemann, 2023-01-09:
You are welcome to send a pull-request.
However, it still needs to work without specifying a build time, so a simple 'mvn install' will return artifacts containing the current timestamp.

Dettorer, 2023-01-20:
Closed in https://github.com/hneemann/Digital/commit/e7cea9edf4a77444507e3a55d8284ca62a183af4.

closed 2023-01-20 by Dettorer