hneemann/Digital #1093: Spanish translation update

hneemann, 2023-01-08:
@angelmicelti
It has been a long time since there was a new release. 
In the next few weeks I plan a new release, and again I would appreciate your help with the translation.

[lang_diff_es.xml.zip](https://github.com/hneemann/Digital/files/10368497/lang_diff_es.xml.zip)

angelmicelti, 2023-01-09:
Hi, happy New Year!
Of course, I'll help you.
Here you are the translation!
Waiting for the new version.
Bye!
________________________________________
Ángel Millán
Profesor de Tecnología <http://angelmicelti.github.io>
I.E.S. Virgen de Villadiego
<https://blogsaverroes.juntadeandalucia.es/iesvirgendevilladiego/>
Peñaflor (Sevilla)

hneemann, 2023-01-09:
@angelmicelti 
Hmm! It seems to me that there is no file attached. 
Maybe email attachments are not automatically attached to the issue?
But anyway, Happy New Year to you too! :smile:

angelmicelti, 2023-01-10:
Sorry, here it is!
[lang_diff_es.zip](https://github.com/hneemann/Digital/files/10383044/lang_diff_es.zip)

hneemann, 2023-01-10:
@angelmicelti 
Thanks again!