hneemann/Digital #1094: Exponentiation Operator?

clifcox, 2023-01-08:
Greetings,

Like so many before me, I was searching for something to replace Logisim, when I found Digital. :-)

I've been learning about generic circuits, and just for fun thought Id try and make a decoder with an enable input. Though I got stuck when I couldn't find an exponentiation operator, which is handy to calculate the number of outputs from the number of selector bits. I thought it might be "**" or a function like pow(), I tried many things. Though I might be missing something, is there an exponentiation operator? :-)

    Thanks,
    Clif

hneemann, 2023-01-09:
If `n` is a positive integer, the easiest way is:
```
2**n = 1<<n
```

clifcox, 2023-01-09:
Thanks,
Clif

closed 2023-01-09 by hneemann