hneemann/Digital #1098: Poor random random number function performance as seeds approch 64 bits

clifcox, 2023-01-14:
Greetings,

recently I was attempting to use 64 bit random numbers to test a multiplexer that split the bits up into 8 bit channels. I was surprised to find out that some of the least significant bits would always be zero, somewhere around 12 of the right hand bits would be zero if you used a seed of random(0x7FFFFFFFFFFFFFFF)) Here is an example of what you might get:

> 6F558D85E8B10000	377CE94A16CFB800	10323E05952B3800	46ECD47D3E3B0800	5860FEB6F1C1F400

The number of zero bits gets less as you lower the seed with the problem just getting started around 0x7F FFFF FFFF FFFF.
Though as you try even larger seeds the behavior gets much more interesting. A seed of all 1s returns 0 every time. All ones except for the LSB, randomly returns either all ones or all zeros, so I guess that's a random true / false feature. ;-) As you count the seed down, gradually more of the LSBs start changing in the results.

Anyway, I'm sure it was optimized for speed, and most people don't need 64 random bits, but it might be a bit confusing if that's what you were expecting. :-) Of course, you could just OR two 32 bit random values together if you knew that's what you needed to do. Anyway, here are my system specifics:

>OS: Linux
>OS version: 5.10.0-20-amd64
>JRE: 11.0.16
>VM: 11.0.16+8-post-Debian-1deb11u1 
>
>Digital.jar
>Build git-Revision: v0.29
>Build Time: 2022-02-11 17:13 

I've also attached more examples results with the seed 0x7FFF FFFF FFFF FFFF.

Thank you  again for a great project!
Clif

[Digital_Random.txt](https://github.com/hneemann/Digital/files/10416793/Digital_Random.txt)

closed 2023-01-14 by hneemann

hneemann, 2023-01-14:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

clifcox, 2023-01-14:
Great,
much better!   But...

Using a seed of 0xFFFF FFFF FFFF FFFF still gives a very non-random zero as a result. :smirk:

hneemann, 2023-01-15:
Java does not support unsigned integers! By now I should know that! :smile:

Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

clifcox, 2023-01-15:
Yay, works great now.  :grinning:  Though I can only check by eye. If only there were unit tests for random functions, probably too much work. ;-)

commit 68cd692b (2023-01-14): fixes an issue in the random number generator used in test cases; closes #1098