hneemann/Digital #1119: Current state at Error

mengstr, 2023-02-18:
It would be great to be able to see the current state of all wires when you get an "Error calculating a step. More than one output is active on a wire, causing a shore circuit".  As it is now only the "shorted" wires are highlighted in red, all other sites turns blue. And to be really useful the error popup should not be modal so you can pan/zoom in the schematic.

Is this return simply overlooked as non-usable, or does the internal design of Digital prevent this functionality?

hneemann, 2023-02-19:
Keeping the wire colors while the error message is shown is not a big deal. 
But making the error dialog non modal is another story. In this case one could do all kinds of things while the model is in an undefined state because of an exception thrown anyware in the model calculation. 
It is difficult to fully understand what this could possibly cause.

mengstr, 2023-02-19:
It seems like I'm guilty of an XY problem here in regards of the modal :) What I really wanted was to be able to zoom and pan with the colors still there after an error, so if the colors could kept also after the model is closed (just as the red wires) it would be even better than making the popup modeless.

But even if the colors are only kept while the modal is open that would be a great improvement. It's just a matter of having the pan/zoom set right when the error occurs.

commit f8b800a6 (2023-03-04): keeps the visual state of the circuit in case of an error, closes #1119
closed 2023-03-04 by hneemann