hneemann/Digital #1128: Better/more highlighting for the Find function

mengstr, 2023-03-11:
The find function is a really nice addition, but it is not particularly good at highlighting the found objects in the schematic - especially when the design is a bit larger. Could you make more visible somehow? Maybe a much larger and colourful blob than the current pale cyan. Or maybe even better - lines drawn from the top of the screen that ends at the objects.

[image]

hneemann, 2023-03-11:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/) and let me know how it feels.

mengstr, 2023-03-11:
Ohh...  That is good!  Fat and, compared to other items, large when zoomed out, and then shrinks to a more normal size when you zoom in so it doesn't block the view.  Now I don't have pan and hunt all over the schematics to locate the found items. Thank you.

commit f6ba0962 (2023-03-11): adds a min size for highlighted components; closes #1128
closed 2023-03-11 by hneemann