hneemann/Digital #1133: SystemVerilog

j054n, 2023-03-22:
Is SystemVerilog support planned in Digital?
Thanks

hneemann, 2023-03-22:
Isn't SystemVerilog an extension of Verilog?

j054n, 2023-06-08:
Originally intended as an extension of Verilog, but more like a superset of Verilog.
The main objective is to integrate modules written in Verilog/SV into the Digital tool with full compatibility with respect to datatypes and parameters in module interface according to the updated syntax of the language.
For simulation purposes I understand that it depends on the external tool configured (Icarus Verilog).

ThanasisVlioras, 2023-07-18:
I find Icarus Verilog is not really good at understanding SystemVerilog. A workaround I found is to use a tool called sv2v. It converts SystemVerilog files to plain Verilog. Try it ! [Github Link](https://github.com/zachjs/sv2v)

j054n, 2023-07-19:
I have tested the tool sv2v you suggest, it works fine, but it doesn't solve the problem I have with Verilog or SystemVerilog.
On one hand the understanding/parsing of Verilog/SV versions with Icarus Verilog can be chosen with -g modifiers:

```
Usage: iverilog [-EiRSuvV] [-B base] [-c cmdfile|-f cmdfile]
                [-g1995|-g2001|-g2005|-g2005-sv|-g2009|-g2012] [-g<feature>]
                [-D macro[=defn]] [-I includedir] [-L moduledir]
                [-M [mode=]depfile] [-m module]
                [-N file] [-o filename] [-p flag=value]
                [-s topmodule] [-t target] [-T min|typ|max]
                [-W class] [-y dir] [-Y suf] [-l file] source_file(s)
```

Moreover, the problem in Digital tool come from parsing the module interface with `parameter` keyword in Verilog/SV, by example:
```verilog
module counter (
	clk,
	reset,
	q
);
	parameter N = 8;
	input wire clk;
	input wire reset;
	output reg [N - 1:0] q;
	always @(posedge clk or posedge reset)
		if (reset)
			q <= 0;
		else
			q <= q + 1;
endmodule
```
and specifically in SV keywords like `logic`, by example:
```verilog
module counter #(parameter N = 8)
                (input  logic clk,
                 input  logic reset,
                 output logic [N-1:0] q);

  always_ff @(posedge clk, posedge reset)
    if (reset) q <= 0;
    else       q <= q + 1;
endmodule
```

I suggest like possible enhancement of Digital tool a parsing of these keyword.
Thanks in advance