hneemann/Digital #1141: Option to disable auto-incrementing labels

OVERKILL27, 2023-04-02:
I am working on a circuit that has a lot of DIP switches. These DIP switches have numbered labels, I need to make a lot of copies of the circuit and need the numbered labels on the DIP switches to remain unchanged in the copies.

In general, the auto-incrementing labels is a convenient feature most of the time, however with the current circuit I'm working on it is forcing me to relabel a lot of DIP switches when all I want is for the copies to have the same labels as the originals they are copied from.

Is there an option somewhere to disable auto-incrementing labels?

hneemann, 2023-04-02:
No, unfortunately there is no such option.

OVERKILL27, 2023-04-03:
Thanks for the quick response Helmut!

I'm really enjoying using this software. It's helped me understand a lot and it really shows that a lot of passion has been put into it!

In lieu of such an option I've resorted to editing the XML .dig files in a text editor and using find and replace as it's faster than decrementing back the labels within Digital but doing that is still a bit of a pain...

This is the only issue that's come up for me while using Digital, would it be difficult to add this option?

hneemann, 2023-04-03:
Please try the latest [pre release](https://infdigital.dhbw-mosbach.de/).

closed 2023-04-03 by hneemann

OVERKILL27, 2023-04-03:
I just tested it. That's exactly what I was looking for! Thank you so much!!!

commit aa9c1427 (2023-04-03): allows disabling of auto-incrementing labels; closes #1141