hneemann/Digital #1146: RAM, Chip Select Multi bit High-Z

esbon125, 2023-04-12:
I was playing with this RAM to make the bidirectional connection immune to short-circuits, so I got this, in which I can read and write correctly and I also can have Z in the output, while CS is high.
[image]
The issue is that after I have enabled the chip but not the output (i should have High-Z in the output), I get weird values. I'd like to ask if it is because drive doesn't take High-Z as an input value or if I'm doing something wrong. 
[image]

                       Thanks in advance, esbon.

hneemann, 2023-04-12:
If a real cmos input is not connected, the read value is more or less random. It is not really random, since it depends on parasitic capacitances and all sorts of hard to predict things. The simulator simulates this behavior by generating a random value when a cmos input is connected to high-z and its value is used nevertheless. See https://github.com/hneemann/Digital/issues/36#issuecomment-315969098 and https://github.com/hneemann/Digital/issues/665#issuecomment-786023684
Thus, outputting a high-z value is not a problem, but reading such a value is.

esbon125, 2023-04-12:
I've read the comments you mentioned and I completely agree with you, I guess I wasn't thinking of them as real components, we could say that I was 'softwaring' the circuit 😅. I found other way around, so I can still do the same, but now I don't have High-Z as input. Thank you for answering.

closed 2023-04-12 by hneemann